RomanMindlin/gpiotest #12: i got this 'Failed to add edge detection' on pi4B 4G

kgbplus, 2020-10-19:
i got this instead on pi4B 4G

```
root@raspberrypi:~# sudo python gpiotest.py  
Failed to add edge detection
```

_Originally posted by @tyjtyj in https://github.com/kgbplus/gpiotest/issue_comments/693561612_

kgbplus, 2020-10-19:
same with raspberry pi3B+, return error 'Failed to add edge detection'

_Originally posted by @RiadySantoso82 in https://github.com/kgbplus/gpiotest/issue_comments/694761322_

kgbplus, 2020-10-19:
Please read [this](https://www.raspberrypi.org/forums/viewtopic.php?t=183481) and [this](https://www.raspberrypi.org/forums/viewtopic.php?t=205327)
Try:
1. Update you Raspberry OS install
2. sudo apt update; sudo apt install --reinstall raspberrypi-sys-mods;sudo reboot
3. Disable 1-wire interface with raspi-config

Gaboris, 2021-01-25:
Sadly this fix did not help in my case on RPI 4 4GB with the latest 32bit Raspberry OS (Buster).

Based on the links even checking `/dev/gpiomem` didn't help since all permissions were right.

kgbplus, 2021-01-28:
Could you please:
1. Provide me image of your current sd card?
2. Try to clean install current version of Raspberry Pi OS and try again?

EdKok, 2021-08-15:
Same error for me as well. RPi4 4GB.
Have multiple Python scripts working with GPIO just fine.

tuctboh, 2023-07-03:
Hi,

Getting this on RPi 4 8G
Kernel 6.1.21-v8+
Debian 11 / Bullseye

Happens when its initializing (initGpio) , in the GPIO.add_event_detect on channel 14.  I have to unload and reload the i2c-bcm2835 module to get functionality back. Otherwise, if I jump that out for GPIO14 things seem to work fine. Why is 14 an issue?

Tnx, Tuc

kgbplus, 2023-07-20:
Maybe it depends on the settings of a serial port which is on GPIO 14, 15? Could you please check?

tuctboh, 2023-07-20:
So the hat I have doesn't mention what GPIO pins it uses, just I2C ports. So I'm not sure what GPIO its using.

mattmartini, 2025-01-24:
I am having the same issue on `Raspberry Pi 4 Model B Rev 1.5` running `Linux 6.6.62+rpt-rpi-v8 - Debian GNU/Linux 12 (bookworm)`