RomanMindlin/gpiotest #16: Running on RPi compute 4 module

neil-benn, 2021-03-18:
Hello,

  I'm trying this on the RPi compute 4 module mounted onto an RPi compute 4 IO board; the argument is b03410 but there are multiple configurations of the compute 4 module.  From what I can understand this combination is the same as an RPi 4.  Is it just a case of adding this into the code to handle the compute module?

Cheers,

Neil

kgbplus, 2021-03-30:
Hello!
I tried to find this revision code but with no luck ( 
Seems, correct revision is b03140 according with that document https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md
Could you please check it one more time?

neil-benn, 2021-03-30:
Actually you are right, sorry.  The RPI compute on a 64 bit setup is giving me a03140!

kgbplus, 2021-04-22:
"Compute Module 4 2Gb" added

closed 2021-04-22 by kgbplus